markh-de/KiVar #14: plugin icon does not work

honzakutil, 2024-02-06:
**Describe the bug**
Windows version of KiCad does not show correct icon for KiVar plugin. It shows general puzzle icon instead

**To Reproduce**
1. install plugin as described in documentation
2. in pcbnew, do: tools -> external plugins -> refresh plugins
3. in pcbnew preferences -> PCB editor -> action plugins select "show button" for KiVar and hit OK
4. KiVar shows in pcbnew top panel, but icon is just general puzzle picture, not picture contained in the plugon repository

**Expected behavior**
Plugin icon will be shown instead of general puzzle icon

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windown 11, tested with KiCad 7.0.10 and 8.0.0-rc2

honzakutil, 2024-02-08:
Icon now work correctly (tested on kicad 7.0.10 and 8.0.0 rc2), but plugin can not be installed according to actual documentation. Inside the plugins folder, KiVar folder must be created, icons and _inint_.py (renamed kivar.py I guess) must be copied.

I propose following fix:
1) move /plugin/resources folder somewhere else in this repository
2) rename /plugin/plugins to /plugin/KiVar ... then the folder KiVar can be copied to kicad plugins folder according the actual documentation

markh-de, 2024-02-08:
Okay, [it's done](https://gitlab.com/kicad/addons/metadata/-/merge_requests/300). KiVar is now part of the official PCM repository. :grin:

Can't wait to see it really popping up [in the official list](https://www.kicad.org/pcm/) (there is some delay).

So, installation instructions will be updated soon (well, there isn't much to say about PCM installation, I'll probably link some KiCad PCM documentation and explain how to enable the toolbar button, just as in the manual instructions).

Thank you for your mail and your motivating words, which have helped me understand that KiVar can really make a change for KiCad users! :+1: