markh-de/KiVar #22: separate options for dnp, not-in-bom and not-in-pos

honzakutil, 2024-02-08:
Why is there only option-!, which controls fit/unfit status of the component? If I understand it correctly, this option control "not in bom", "not in pos files" and "dnp" kicad attributes together. Did you think about adding separate options for controlling the attributes individually (somethink like -dnp, -nbom, -npos + keeping the -! as shorthand to manipulate all of them at once)?

There is one marginal example, of usage separate options:
- lets suppose, I have one PCB design, but there is more variant of bare pcb, which can be produced from the design (different copper thicknesses in pcb stackup, different layer count)
- it is good idea to add PCB itself to the BOM file (just as reminder for assembly house, that they needs to order pcb alng with other components)
- so I would create "virtual" componnet to appear in BOM
- since there is multiple variants of the board, I would use KiVar to take care of the variants
- there is no need to have entry for PCB in position files

I know the example above is really unlikely to happen, so take it as interesting thing to think about. The truth is, that I met with all listed conditions in real life. Having PCB as one item of the BOM was requirement of one assembly house and I already designed several boards, which use same gerber data, but different fabrication specs.

markh-de, 2024-02-09:
It seems you already read [my notes in the source code](https://github.com/markh-de/KiVar/blob/27efd180899610e3d9919a96dc18f6a4c35c5441/plugin/de_markh_kivar.py#L90). :smile:

Yes, I have already thought about separate options, but I wasn't sure if it really covers a usecase.

I was more thinking about a (board specific) config menu where users can define what the `-!` option really does (set/clear/keep each of DNP, nPOS, nBOM), because I thought that I wouldn't make much sense to treat components differently (e.g., either you want unfitted parts in your BoM or you don't - this is not a component-specific decision). The idea was that you mark unfitted parts with `-!` and then later decide what the `-!` really does, depending on the requirements of your assembly house, for example (whether they want unfitted parts in the BoM, with a "DNP" column of course).

Admittedly, I'm not sure how such a feature would help with your PCB-in-BoM issue, because I have not yet understood which attributes of your virtual PCB "component" you wish to automagically change via KiVar. Could you please provide an example for 2..3 different cases?

Anyways, this feature would be possible. Let's discuss use-cases before we start implementing it.

honzakutil, 2024-02-09:
Yes, you are right, I looked also at the code and notes there.

I realized, that usecase with pcb will not require to have separate options of dnp, nbom, npos. KiVar will be needed only to change value of pcb component and leave options unchanged (I believe that is what kivar do - it does not change actual state of dnp, npos and nbof, if it is not istructed to do it by -! option).

One think I like about KiVar is that it allows me to store information about assembly variant inside the KiCad and a do not need to keep some tables somewhere else. For components affected by KiVar I always show the Kivar.Rule textfield in the schematic. It should remind me, that this component is somehow special and a need to be careful when making changes. If there will be some hidden, board-specific config, it would not be so obvious, what is going to happen. This hidden config could be also tricky, when copying part of one project to another one and projects will have different KiVar config.

If independent options for dnp, nbom, npos  will be used, storing KiVar configuration will not be needed. Except of this, I do not see any other usecases for separate options right now.

markh-de, 2024-02-09:
> I realized, that usecase with pcb will not require to have separate options of dnp, nbom, npos. KiVar will be needed only to change value of pcb component and leave options unchanged (I believe that [...]

Well, unfortunately, you're wrong. In its current implementation (and concept), KiVar _requires_ (for choice matching) and _sets_ (when applying choices) values and attributes in/to a clearly defined state. It does not allow a mixture of manually and automatically assigned attributes. As documented, when you do not use `-!` the part _will_ be fitted (hence all related attributes, i.e. dnp/npos/nbom, _will_ be cleared, not kept in their previous state). Vice versa, if `-!` is used, the part _will_ be unfitted and all related attributes _will_ be set.

So you want to let KiVar handle a component, but you want to leave the dnp/npos/nbom attributes in their original state. In its current state this is not supported.

For this to work, we would need to either
* introduce a rule-related option (not a choice-related option, because disabling attribute-matching is not choice-specific), or
* implement the separate options as originally discussed and use them to define the attribute state of your virtual PCB component.

markh-de, 2024-03-11:
Along with #26, this is now implemented. There is no real documentation for it yet (check the commits, I added some very short info to 439c20782e586fd1a8a0e168f683fc9eaa3e0e61), however, at least the demo project has some updated rules (although I commited only chunks, so maybe sch and pcb are not in sync ... I need to call it a day now).

I'll update the documentation soon.

closed 2024-03-11 by markh-de