markh-de/KiVar #24: control KiVar from command line

honzakutil, 2024-02-08:
Is it possible to control KiVar from command line? I am using KiBot to generate gerber data and position files. If I could call kivar from command line and tell it somehow, which variant I need to set, I can automatically create position files for all assembly variants by running single script which would do following: 
- call kibot to generate gerbers
- call kivar to set variant 1
- call kibot to generate position files for variant 1
- call kivar to set variant 2
- call kibot to generate position files for variant 
- ...

markh-de, 2024-02-09:
Well, I didn't make a note about this in the source code, did I? So I guess you've read my mind this time. :smile:

Yes, it was clear to me that users will ask for a CLI soon. As I'm a fan of automation, too, I think this should be done.

It'll be a bit of code rework and separation, but technically it should be no problem.

Until I find the time to work on this, let's use this ticket for feature discussion.

Some call examples which will explain the kind of features I have currently in my mind:

`kivar_cli --list example.kicad_pcb`

* Lists components (only those with rules assigned) along with their value and attributes for each choice (already "flattened", i.e. without default choices, but dedicated values/attr for each possible choice as they would really be assigned).

`kivar_cli --state example.kicad_pcb`

* Presents a list of aspects along with each currently matching choice (or "unset").
* If `--all` is added, inactive choices are also listed, so that the user can see what else is possible for each aspect. The currently selected choice could be marked with a `*`.

`kivar_cli --set ASPECT1=FOO --set ASPECT2=BAR example.kicad_pcb`

* Assigns choices to aspects (can be only a subset of possible aspects). Details about call syntax TBD (multiple `--set` options or only one with a list of assignments ...).

... and so on.

Comments and ideas are welcome.

honzakutil, 2024-02-09:
I agree, `--list`, `--state` and `--set` will be the most useful commands. Maybe it will be good to add another command:
- `kivar_cli --check example.kicad_pcb` - simply return error, if KiVar is unable to resolve actual choices for all aspects (= any <unset> is found). If option `--verbose` is added, KiVar will also print a list of components, which are causing the inconsistency

Maybe it will be better to use syntax similar to git command (`git log --oneline --graph origin/master` - `command subcommand --arguments target`). Commands could look like this:
- `kivar list example.kicad_pcb`
- `kivar state --all example.kicad_pcb`
- `kivar check --verbose example.kicad_pcb`

Having possibility to use kivar from commandline could also solve the problem with modifying text variables from kivar in order to  get some text string and use it for name of output file produced by KiBot. Output of `kivar state` could processed by custom script and saved in enviroment variable, which could be then used by KiBot for output files naming.

markh-de, 2024-02-09:
I like your suggestions.

A `check` command sounds useful, and using a mandatory command argument instead of options makes perfect sense. :+1:

Also, the commands that perform any changes to the board shall have a `--dry-run` option.

honzakutil, 2024-02-13:
nice, I will try it as soon as I have a little time for it.

First real application for this could be semi-automatic test of KiVar itself. There would be one set of kicad PCBs which will be passed as argument to kivar. After kivar processes the board, it will be compared (e.g. using `git diff`) to the result board, where correct state was set and checked before. I will try to prepare these test/correct board pairs and some test script. If this works, next step could be fully automated test using CI/CD pipline.

markh-de, 2024-02-13:
The `list` command is now working as described [above](#issuecomment-1935527102).

Using the CLI for testing is a good idea. For this purpose, I have just added a `--query <ASPECT>` option to the `state` command, so that you wouldn't need `git diff` to check the matching choice for an aspect. In query mode, only the unformatted name of the matching choice is printed (empty line and exit code 0 if no match found, exit code != 0 if aspect name was not found).

PR #39 (2024-02-23): Add Command Line Interface application
This splits the Action Plugin into the plugin and a module. The module is also used by the CLI app.
The CLI app is not yet part of any distribution package.
To be able to work on other stuff without diverging too much, it's time to merge to main branch.

Fixes #24
closed 2024-02-23 by markh-de