markh-de/KiVar #26: assembly variants for other fields than value

honzakutil, 2024-02-13:
According to the plugin documentation, there is way how to vary multiple component fields by having multiple instances of the component and switching `-!` options to fit the component with desired set of fields. Downside of this approach is more complex schematic and overlapping footprints, which require some drc tweaking. Here is idea of another possible approach:
- instead of `KiVar.Rule` field for variatlion of the `Value` field, lets use `KiVar.Value` field
- for variation of other fields, lets use `KiVar.<fieldname>` field
- all `KiVar.<fieldname>` fields for one components needs to be compatible. Easiest way (and the least user friendly) to achieve this is to require them to have same *aspect name* and same *choice names*

Here is example for variation of power transistor (value, datasheet, description):¨
- `KiVar.Value` = `VOLTAGE 80V(BSC026N08) 100V(BSC035N10)`
- `KiVar.Description` = `VOLTAGE 80V('80V Power transistor in TDSON8 package') 100V('100V Power transistor in TDSON8 package')`
- `KiVar.Datasheet` = `VOLTAGE 80V('http://...link-to-bsc026.pdf') 100V('http://...link-to-bsc035.pdf')`

Another example is variation of value and LCSC partnumber for pull-down resistor:
- `KiVar.Value` = `pulldown weak(100k) strong(1k)`
- `KiVar.LCSC` = `pulldown weak(C25803) strong(C21190)`

markh-de, 2024-02-26:
This feature is now implemented on my local repo (still heavily WIP). I'll push it to a new branch within the next days.

The concept:
You still have the main rule (`KiVar.Rule`) as you know it, but you can now specify so-called **Auxiliary Rules**, which do not _define_ choices, but can only _refer to_ choices that are declared in the main rule.

The main rule stays special for these reasons:
1. It defines the aspect (aux. rules must use the same aspect that is specified by the main rule).
2. It is the only rule that declares/defines choices (aux. rules only refer to these choices, but do _not_ introduce new ones).
3. It always defines the component value (cannot be redirected to a random field).
4. It defines the component attributes (DNP, nPos, nBoM, ...) for each choice.

Because of these special functions, it will keep a special name (currently `KiVar.Rule`), while the aux. rules are currently (maybe subject to change) found under `KiVar.Aux`.

For your above examples the usage would be:

- `KiVar.Rule` = `VOLTAGE 80V(BSC026N08) 100V(BSC035N10)`
- `KiVar.Aux.Description` = `80V('80V Power transistor in TDSON8 package') 100V('100V Power transistor in TDSON8 package')`
- `KiVar.Aux.Datasheet` = `80V('http://...link-to-bsc026.pdf') 100V('http://...link-to-bsc035.pdf')`

Side note: Another new feature is that you can now (optionally, and mutually exclusively) define the aspect name in a dedicated field `KiVar.Rule.Aspect`, instead of declaring it as the first "word" in the `KiVar.Rule`. Reasons for this addition:
1. People may not always want to show the aspect name as part of the choices when the rule field is shown next to their symbols. Optionally moving the aspect name to a dedicated field saves space in the main rule.
2. It makes the style of the main rule and the aux. rules similar, in that they may only contain the choices, not an aspect name.

If you choose to use the new aspect field, your above example would become:

- `KiVar.Rule.Aspect` = `VOLTAGE`
- `KiVar.Rule` = `80V(BSC026N08) 100V(BSC035N10)`
- `KiVar.Aux.Description` = `80V('80V Power transistor in TDSON8 package') 100V('100V Power transistor in TDSON8 package')`
- `KiVar.Aux.Datasheet` = `80V('http://...link-to-bsc026.pdf') 100V('http://...link-to-bsc035.pdf')`

You will receive an error if you try to use the new aspect field and specify the aspect as the first "word" in the rule specification, so you must decide which style you want to use (per-component).

I'm still not sure about the new field names. Maybe it'll become `KiVar.Aux(Datasheet)` or `KiVar.Aux:Datasheet` ... I don't know yet. KiCad component fields seem to be case-insensitive (you cannot have "foo" and "Foo" at the same time), so I'll make the KiVar fields insensitive, too. However, thinking forward, there may be parts in the field names where the capitalization might matter, so we'd need a syntax which makes it clear where the insensitivity stops. E.g. everything inside parentheses (or after a colon) could be "protected". E.g. `kivar.somefeature(FooBar)` would be equivalent to `KiVar.SomeFeature(FooBar)`, but not to `KiVar.SomeFeature(foobar)`.

markh-de, 2024-03-02:
There is a feature branch available for testing now.

I'll merge this to main within the next days to avoid diverging branches when I start implementing other stuff (e.g. #22).

markh-de, 2024-03-11:
Along with #22, this is now implemented. There is no real documentation for it (check the commits, I added some very short info to 439c20782e586fd1a8a0e168f683fc9eaa3e0e61), however, at least the demo project has some updated rules. I'll update the documentation soon.

closed 2024-03-11 by markh-de