markh-de/KiVar #27: kivar_cli: instalation and usage on windows

honzakutil, 2024-02-15:
I am using KiCad on regular windows machine, so I tested `kivar_cli` there. Here are my observations:
- just mention for the beginning: kicad installation on windows has its own command prompt, it is not possible to use kicad from regular command line or powershell
- `./kivar-cli --help` does not work for windows, it probably does not uderstand the `./` and the fact, that it is python script. I was able to run it by command `python kivar-cli --help`
- it is possible to run the script only from the folder, where it is located. It should be probably added to $PATH or something like this during plugin installation
- author of [KiKit](https://github.com/yaqwsx/KiKit) plugin was probably solving similar issue. According to its [documentation](https://yaqwsx.github.io/KiKit/v1.4/installation/intro/), KiKit has multiple parts - action plugin for pcbnew (installed by PCM) and *backend* installed from kicad command prompt using `pip`.

markh-de, 2024-02-16:
I guess that most of the things you've mentioned are not specific to KiVar, but general Python runtime issues on Windows. If I were to use Python on Windows, I'd use WSL (but that's just my personal preference).

Please note that it's not planned to provide the CLI as part of the plugin package. Instead, it will probably be distributed independently via PyPI (`pip`).

As you mentioned, KiKit does it the same way. First, because KiCad Action Plugins cannot have dependencies (that's a real pity, but I understand how complicated things would get with dependency management). And second, because you don't want to force users into having to run their CLI script from within your KiCad release's plugins directory (or create a symlink to it). There may be even more reasons.

honzakutil, 2024-02-17:
I found only one practical example, when I would like to use `kicad-cli` on windows:
- `kivar-cli list --detailed board.kicad_pcb` - I really like to see a list of componnets affected by given aspect. This was inspiration fof feature request #38 

All other features all accessible via gui, so there is no reason to use `kivar-cli` during board design process. Two main usecases for `kivar-cli` I can think of are:
- automated testing of `kivar` itself
- usage during automated fabrication data generation to set given assembly variant or to check, if current board state is valid

It is likely, that both usecases will use kicad on linux, maybe [kicad_auto](https://github.com/INTI-CMNB/kicad_auto). Maybe it is not worth the effort to make `kivar-cli` to run on windows, perfectly.

markh-de, 2024-02-27:
Related to #40

honzakutil, 2024-05-22:
As I reported in #62, it is possible to install and use kivar cli using pip from the locally created python package.

closed 2024-05-22 by honzakutil