markh-de/KiVar #28: omit board filename if only one is in the current working directory

honzakutil, 2024-02-15:
To save some typing, it could be beneficial allowing to ommit pcb filename, if only one pcb file is found in current working directory. This would be usually the case, since each kicad projects resides in its own directory and usually have only one pcb file. If multiple files with pcb are found, kivar should print an error message.

markh-de, 2024-02-16:
Hm, I'm not sure about this one.

I understand what you mean. However, having grown up in a Linux world, I'm a bit against this kind of 'I know what you mean' mechanism. It's just not typical for a CLI tool. If you want the tool to work on your specific file, you'll need to tell it which file you mean.
If you want to type less, you can use your shell for resolving the file name. E.g., use `*.kicad_pcb`, which will resolve to one file name, if there is only one file (and if there are more than one, the app will complain about too many arguments), or use file name completion, or write a wrapper script.

I'd say this kind of usage is a question of personal taste and should rather be handled by the caller outside the CLI app. I'd like to avoid guessing what the user meant, and I'm not aware of any other CLI tool on Linux working this way.

I'll keep the ticket open and think about it. :smiley:

honzakutil, 2024-02-17:
I agree it is about personal taste. Another reason why not to do that could be, it is just not worth it. Typical use of `kivar-cli` will be probably as a part of some script, hence it woult be typed only once. It would save a typing only for manual testing and playing with `kivar-cli`.

However, at least one tool with such behavior exists - [KiBot](https://github.com/INTI-CMNB/KiBot). I am using using it inside the docker to generate all the data from my designs. I just start the docker container, `cd` into the project directory and type `kibot`, which does all the magic. It seems, that `kibot` is not competely happy with this guessing what to do - it prints the warning. Btw, the `kibot` was my original inspiration for this ticket.

markh-de, 2024-02-18:
OK, so let's keep the usual behavior and require the file name.