markh-de/KiVar #29: kivar-cli: syntax error of KiVar.Rule does not report an error

honzakutil, 2024-02-15:
Originally, I had following KiVar rule: `VOLT_RANGE 1208(BSC026N08NS5) 1210(BSC035N10NS5) 1308(ISC014N08NM6) 1310(ISC022N10NM6)`. In order to test KiVar, I deleted first closing parenthesis, so the rule was: `VOLT_RANGE 1208(BSC026N08NS5 1210(BSC035N10NS5) 1308(ISC014N08NM6) 1310(ISC022N10NM6)`. When I tried to run KiVar plugin from gui, it reported error, as expected: *Q3: rule splitter error: Unmatched opening parenthesis*. When I tried `kivar-cli` with pcb containing this syntax error, no error was reported, `kivar-cli` worked "normally". According to output of `kivar-cli list -d board.kicad_pcb` it was just ignoring the part Q3, which had syntax error in the KiVar.Rule field. I would expect the `kivar-cli` to print same error, as the gui.

markh-de, 2024-02-16:
You must not forget that the CLI app was hacked together on one evening. It's totally WIP, that's why it's not yet merged to the main branch. The CLI branch is more or less a playground for you. :smiley:

Yes, [I'm aware that error reporting does not yet work](https://github.com/markh-de/KiVar/blob/add-cli/plugin/kivar-cli#L8). I'll soon add a wrapper to the rule parser and care about printing error messages there.

markh-de, 2024-02-17:
Just added the wrappers. Errors are now printed.

Also, the file name is now checked first, so that you don't get a scary call stack from within `pcbnew.LoadBoard()`.

Please report whether it works for you.

closed 2024-02-17 by markh-de

honzakutil, 2024-02-17:
Yes, it works, error is printed now. I tried the same trick with missing parenthesis and I received expected error message.

If `kivar-cli` was hacked together in one evening, it is pretty impressive, that it worky and there are only few minor bugs. Good job!

PR #37 (2024-02-17): Add error reporting
Fixes #29