markh-de/KiVar #41: Incorrect RefDes when reporting rule error

honzakutil, 2024-03-05:
Kivar should report error `Rule mixes choices with defined (2) and undefined(1) values (either all or none)` for components R13 and R15 in attached pcb. Instead of that, it reports the error multiple times for R18 (last component checked). When the last line of report is clicked, R15 is correctly highlighted, if previous line is clicked, the R13 is correctly highlighted. It looks like error was identified correctly, but refdes was accidentally overwritten by the refdes of component, that was checked at the very end.

Tested with KiCad 8.0 and KiVar 0.2.0-dev6 on machine with win11

**example PCB:**   [errors.zip](https://github.com/markh-de/KiVar/files/14495867/errors.zip)

**R13 selected:**  
[image]  

**R15 selected**  
[image]

markh-de, 2024-03-05:
Hi. I'm short of time. Please check 0.2.0-dev7. Thanks!

honzakutil, 2024-03-05:
Thanks for quick reply. 0.2.0-dev7 solved the issue.

markh-de, 2024-03-05:
Yeah, sorry for the oversight. I recently changed the way footprints are referenced in the internal database. In the past, the refdes was the main key in the database. This prevented you from assigning rules to footprints with an identical refdes. Now we use the footprint's UUID and resolve this to the ref later. And there one lookup was still missing.

commit e77a8c1f (2024-03-05): Update ref for error reports

Fixes #41
closed 2024-03-05 by markh-de