markh-de/KiVar #44: kivar-cli ends with error `TypeError: argument of type 'NoneType' is not iterable`

honzakutil, 2024-03-11:
`kivar-cli` ends with `TypeError: argument of type 'NoneType' is not iterable` when run with [this](https://github.com/markh-de/KiVar/files/14556475/errors.zip)  PCB file.
exact command: `python kivar-cli check errors.kicad_pcb` (run on win 11 from kicad command promt in `...\Documents\KiCad\8.0\scripting\plugins`)
kivar version: 0.2.0-dev10
full error message:
```
C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins>python kivar-cli check errors.kicad_pcb
Traceback (most recent call last):
  File "C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins\kivar-cli", line 258, in <module>
    sys.exit(main())
             ^^^^^^
  File "C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins\kivar-cli", line 246, in main
    if not check_command(args.board): # , args.verbose
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins\kivar-cli", line 135, in check_command
    vardict = get_vardict(b)
              ^^^^^^^^^^^^^^
  File "C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins\kivar-cli", line 35, in get_vardict
    vardict, errors = kivar.get_vardict(board)
                      ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins\kivar.py", line 356, in get_vardict
    return build_vardict(load_fpdict(board))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins\kivar.py", line 482, in build_vardict
    error_str = add_choice(vardict, uuid, aspect, name, content, field, all_choices[aspect])
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins\kivar.py", line 308, in add_choice
    return f"Undeclared choice '{choice}' (in aspect {quote_str(aspect)})."
                                                      ^^^^^^^^^^^^^^^^^
  File "C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins\kivar.py", line 115, in quote_str
    if any(c in str for c in ', -\\()='):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\JanHladík\Documents\KiCad\8.0\scripting\plugins\kivar.py", line 115, in <genexpr>
    if any(c in str for c in ', -\\()='):
           ^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable
```

With my poor knowledge of Python and a help of ChatGPT, I was able to fix it by adding following into the function `quote_str(str)`:
```python
    if str is None:
        return "''"
```

Whole function after fix looks like this:
```python
def quote_str(str):
    if str is None:
        return "''"
    if str == '': result = "''"
    else:
        if any(c in str for c in ', -\\()='):
            result = "'"
            for c in str:
                if c == '\\' or c == "'": result += '\\'
                result += c
            result += "'"
        else: result = str
    return result
```

I am not sure, if this is real fix or just workaround, please do a real fix :-)

markh-de, 2024-03-11:
Thanks for testing and reporting. I'll check this.
From the error message it seems that the aspect name is undefined, which should not happen.
I'll try to reproduce this with my latest development version and fix it if it still happens.

markh-de, 2024-03-11:
Good catch. This happened when there were aux rules, but no main rule. Then the aux rule was handled, which it should not.

This is fixed on the `aux-rules` branch.

Thanks for reporting.

closed 2024-03-11 by markh-de