markh-de/KiVar #52: footprint visibility when populating/unpopulating

PerennialNovice, 2024-04-23:
Is it possible to completely remove a not populated footprint from the board layout when switching aspects?

This would be handy for creating different stencils when the choise is between LQFP and QFN package for the same component

[image]

markh-de, 2024-04-23:
Enabling/disabling (i.e. adding/removing) footprints is beyond the (current) scope of KiVar, which really only takes care of _assembly_ variants. The PCB layout (including paste/stencil data) is supposed to be identical between all those variants.

Technically, it should be possible to remove a footprint, but adding it back, along with any user customizations applied to it, is definitely not part of KiVar's intended feature set. Sorry about that.

I hope you find a tool that suits your use case better.

closed 2024-04-23 by markh-de

PerennialNovice, 2024-04-24:
allright, I see I wasn't aware of the scope of this request....
thanks for clarifying :)

markh-de, 2024-05-01:
I just ran into the problem of having to manually enable/disable solder paste for a footprint.

I did this by switching the _Solder Paste Ratio_ of the footprint between 0 and -100%, which worked fine (except for "SMD Aperture" pads, which I had to change to "SMD").

After doing this a dozen times (for exporting Gerbers for various variants) I changed my mind about the scope :smile: and I think this can be done by KiVar. We just have to use a "trick" (some might call it "hack") to save the original value that the user applied.

There is a new issue #53, which you might be interested in.

markh-de, 2024-07-13:
Version 0.3.0 has just been released, and it allows enabling/disabling solder paste per footprint.

You might want to give it a try.