markh-de/KiVar #56: kivar-cli: Script does not read KiVar.Rule

bemmbix, 2024-05-06:
**Describe the bug**
`kivar-cli list Board.kicad_pcb`
returns
`Error: No rule definitions found.`
when field `KiVar.Rule` is used.
e.g. KiVar.Rule = ETH *,NO(-!) YES(+!)

When naming the field `Var` (like done in the Demo-Project) the script returns correct data:
`ETH: NO YES`

BUT... the action script inside KicadPCB doesn't find any data with variabl "Var"...

**Expected behavior**
KiVar.Rule is used from CLI and GUI (not Var)

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: [f5b5434](https://github.com/markh-de/KiVar/commit/f5b5434daa6e79ac90738844e36f411b230e3a61)

markh-de, 2024-05-07:
`KiVar.Rule` is the v0.1.2 field name.
`Var` (plus some more) is the new v0.2.0 field name.

It seems you are using the CLI and demo project from the `main` branch, which are not supposed to be used by end-users.

The _main_ branch contains the (unstable, i.e. not settled) development state. I'll do a pre-release of the v0.2.0 CLI on PyPI (i.e. `pip install kivar`) soon (the implementation is alright, but I'm struggling with finding enough spare time to complete the documentation).

For now, please stick with the v0.1.2 plugin release and the demo project and readme contained in its [source archive](https://github.com/markh-de/KiVar/archive/refs/tags/v0.1.2.zip).

I'll also pre-release the plugin, but only as ZIP for manual installation. I don't want to release something in the PCM that is yet missing proper documentation.

For the v0.1.2 syntax there is no released CLI (you'd have to find the first occurrences of the CLI code here in the repo and try it out - it was based on v0.1.2 - but I'd recommend you to wait for the v0.2.0 pre-releases).

You have to be careful with the rule syntax. The v0.1.2 and v0.2.0 look similar, but they have some different interpretations. I see you're using the `+!` property setter. This will be (mis)interpreted by the v0.1.2 (`KiVar.Rule`) as a value to be assigned. The rules are not fully compatible. I'm planning to add a _migration_ section in the readme.

markh-de, 2024-05-07:
I'll close this issue, because it's nothing to be worked on (except for the releases :smile:).

If you have questions about the usage or creating correct rules, feel free to open another issue. I'll tag it as _question_, and we can discuss your use-case there, if you want.

closed 2024-05-07 by markh-de

bemmbix, 2024-05-07:
Thank you for your fast and competent help. I'll wait for version v0.2.0 and then test the plugin again.

markh-de, 2024-05-07:
Thanks for pointing out the potential confusion of releases and the files on the main branch ... admitted, you're not the first one to stumble upon this.

I'm planning on creating a maintenance branch for each minor release, so documentation and demo project can live there in a release-specific context (#58).

Also I just had the idea of creating a pre-release PCM repo (#59). If you are interested in being an early adopter, you can subscribe to this issue to stay informed about progress.