markh-de/KiVar #62: How to install KiVar from actual repository state

honzakutil, 2024-05-09:
How should I install KiVar, if I want to try the newest development version (actual state of the `main` branch)? There are only instructions for PCM and for manual instalation from the release zip archive in the `readme.md`.

markh-de, 2024-05-10:
There is a simple `release` shell script that takes care of generating
 * a KiVar CLI app Python package,
 * a KiVar Action Plugin PCM package,
 * a "classic" Action Plugin ZIP archive.

For Ubuntu (actually, any Linux distro in general, or WSL):
Download the repo (clone or get a ZIP archive), go to the `source` folder, run `./release.sh 0.1.9001` (0.1.9001 being some "close to 0.2.0" version number ... we can only use _major.minor.micro_ scheme).

If it succeeded, you'll find three files in the `../release/0.1.9001` folder:

1. The package with `pcm` in its name is an Action Plugin PCM package, that can be installed with the "Install from file..." option in the KiCad PCM.

2. The `.tar.gz` file is a Python package containing the `kivar` CLI app, to be installed with `pip install <package_file>`.

3. The other `.zip` (without `pcm`) is the "classic" archive containing the Action Plugin, to be extracted into your KiCad plugins directory.

If you fail to create/install the packages on your system (the release script is quite simple and does not yet contain proper error handling), just give me a ping. I already thought about creating a pre-release here on GitHub, so that everyone can give it a try, even before I found the time to complete the documentation.

honzakutil, 2024-05-13:
I tested the results of the `release.sh` script with following results:
- classic action plugin zip archive: does not work - after copying content of the zip into the plugins directory and refreshing the plugins, the KiVar does not show among other external plugins installed in kicad. I guess there is some problem with file naming, but I have not been able to solve it myself. Please take it as note, that actual state of the source does not match the actual state of the documentation. According the other opened issues I understood, you are working on it already :-)
- PCM package: working correctly (open PCM, use install from file ...)
- python package: working. I was able to install it in Kicad command prompt (I am running Kicad on windoew 11) by command `python -m pip install ./kivar-0.999.tar.gz` and then rin it just by typing `kivar`

markh-de, 2024-05-13:
Thanks for reporting.  I would classify PCM and pip the "critical" parts.  Good to hear that they are working for you, even on a different platform than I use (native Ubuntu 22.04).

I'll have a look at the ZIP file when I find the time.

markh-de, 2024-05-13:
Oh, and there actually _is_ a small plugin debugging aid in KiCad:

In the pcbnew preferences dialog, in the Action Plugin section, check if there is a warning icon next to the sorting and file browser icons below the plugin list.  If so, click on it to see what prevented the plugin(s) from loading (stack trace).

markh-de, 2024-05-13:
The classic ZIP should be fixed now. Please test the PCM and Plugin ZIP again (they share code).

Ref: 2cc1a684bb4931d46281c40d0dd58195dd524857

honzakutil, 2024-05-14:
Fix is working, I was able to install all 3 outputs of the release script and start kivar. It reports version 0.2.0-dev32. Tested with latest main (18442f3) on win11 and kicad 8.0.2

closed 2024-09-13 by markh-de