markh-de/KiVar #64: Can I have any attribute change by the selected variant?

rob-bits, 2024-05-14:
I have MPN attribute that has a value for a specific resistor, capacitor. If I change a value, I would like to change the MPN attribute as well. 

Some component has MPN1, MPN2, MPN3 ..etc. I would like to have control on these too. 

There is an attribute SPECS, which contains specification for a component. E.g 10k/0402/1% or 100nF/0603/10%/50V/X7R ..etc. And if the variant use a different value, the specs should be updated too.

How can I do this in KiVar?
Can I add multiple attributes into the Choices?
E.g.:
**Variant1 Test1(Value=3k,MPN=xxcvcx,SPECS=3k/0402/1%) Test2(Value=1k,MPN=yyyxy,SPECS=1k/0402/5%)**

Thanks!

rob-bits, 2024-05-14:
Thanks! I will try this out soon.

rob-bits, 2024-05-14:
Tested, and works perfectly! Thanks!

I used the following example values:

|Field | Value|
| --- | --- |
|Var| Var1(3k) Var2(1k) Var3(DNP -!) R1_aspect|
|MPN.Var| Var1(MPN1Value) Var2(MPN2Value) Var3(None)|
|SPECS.Var| Var1(3k/0402/1%) Var2(1k/0402/5%) Var3(None)|
|Value| DNP|
|MPN| None|
|SPECS| None|

[image]
[image]
[image]

markh-de, 2024-05-14:
Nice.

If you want to assign an empty string in the DNP case (might look cleaner than "None"), just assign an empty quoted string, such as `''` or `""` (support for double-quotes is new). Example: `Var3('')`.

Regarding the alignment of your displayed fields: Try pressing `O` on your component, so that all fields are aligned automatically. Currently, MPN and SPECS seem to be right-aligned (or rotated).