markh-de/KiVar #65: KiVar warn about legacy rules only if there are no rules in new format

honzakutil, 2024-05-22:
Right now, KiVar prints warning about presence of the rules in legacy format (nice feature). It does it only if there are no rules in new format. If both formats are present, no warning is displayed. Maybe it will be goot idea to show this warning always, when legacy format rules are found.

markh-de, 2024-05-22:
Hm, I understand your point.  But the pure presence of additional fields (maybe the design is supposed to be compatible with old _and_ new versions?) should not prevent the user from using newer fields.

The dialog you're seeing is there to tell users that they need to adapt their old rule strings and to point them to the migration guide (to keep the level of confusion and frustration low).  Once they know that there is conversion work to do and where to get help (docs), it's up to the users to decide which rules they convert and which rules they delete.

I know what you mean and I had exactly the same thing in mind.  But I decided that additional rules (that are irrelevant for the new release) should not stop you from using the tool or nag you with a warning screen.  Maybe they're still there on purpose.

markh-de, 2024-05-22:
I think a subtle hint is okay.  The "Usage hints" link and text now change in case legacy rules were found.

Printing text in red may conflict with the GUI theme.  I think default color is good enough.

Please close the ticket if you're okay with that.

Thanks for your input!

honzakutil, 2024-05-23:
nice, thanks!

closed 2024-05-23 by honzakutil