markh-de/KiVar #73: Action Plugin doesn't work on macOS (13.6.9)

Cocoader, 2024-08-21:
**Describe the bug**
Clicking on the action plugin in the pcb editor makes KiCad become unresponsive. 
It doesn't matter if you click the "Close" or "Update PCB" button.
KiCad behaves after that like the "Variant Selection" window is still open somewhere in the background and you can't click anything inside KiCad and have to force quit the application.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on the KiVar Action Plugin in the Status Bar.
2. Select any assembly variant - this is optional and doesn't change the behaviour.
3. Click Close or Update PCB.
4. KiCad becomes unresponsive.

**Expected behavior**
You can select an assembly variant.

**Desktop (please complete the following information):**
 - OS: macOS Ventura 13.6.9
 - KiCad 8.0.4 with bundled Python 3.9.13

**Additional context**
I did try it in a Windows 10 VM with KiCad 8.0.4 with bundled Python 3.11.5 which worked.
I don't understand why KiCad is still bundling such an old version of python and doesn't use the same version for Windows and macOS.
Hopefully you can reproduce this in 3.9.13.

Cocoader, 2024-08-22:
Sorry for omitting the version of KiVar.
I tested it with the **latest** version in PCM (0.3.0) and 0.2.1. Both have a similar behaviour.

Yes, I would gladly help testing. I did look into a couple of other plugins - mainly KiBot or KiCost - and I like your straight forward approach the most.

markh-de, 2024-08-22:
Well, I remembered wrong.  The dialog already was modal, but I `Destroy()`ed the dialog from within an event handler.  I've modified that part.  Please install the following PCM Package using the "Install from File..." button in PCM and give it a try:

[kivar-0.3.9908-pcm.zip](https://github.com/user-attachments/files/16717325/kivar-0.3.9908-pcm.zip)

Please report whether and how it changes the behavior.

Thanks for your help!

Cocoader, 2024-08-22:
Wow - that was a lightning fast response.

I did try your demo project and every example supplies plausible results and KiCad works as expected now.

With 0.3.0 under Windows there where in two or three of the examples (if memory serves me right) some issues which you have fixed with this release as well.

Thank you very much for the rapid fix and I really appreciate your work.

markh-de, 2024-08-22:
Thanks for testing and for the positive response!

Just being curious ... what other fixed issues do you mean?  Were they related to the GUI (I fixed a bunch of graphical bugs) or to the variation data?

Cocoader, 2024-08-22:
>Thanks for testing and for the positive response!

No problem whatsoever. I will follow your progress on this plugin eagerly and report if I find something else.

>Just being curious ... what other fixed issues do you mean? Were they related to the GUI (I fixed a bunch of graphical bugs) or to the variation data?

When the plugin didn't work on macOS I wasn't sure if it was related to KiCad or the bundled python version so I tried it in a Windows 10 VM. Which looked similar unpromising at first.

[image]

But after deleting example 3, 5 and 7 the basic plugin functions worked - no unresponsiveness.
Also without knowing how the plugin would look like working properly I wouldn't have doubted at first that the PCB Assembly Variant Code has been generated incorrect. So I was happy to test the functionality, get a bit more familiar with the plugin and isolate it to the macOS port of KiCad.

[image]

Those errors might have been linked so I would've opened up another issue after you fixed the macOS related issue if they still persisted.

markh-de, 2024-08-23:
You used the demo project from the main branch in a release (=older) version of KiVar.  I know because example 7 is only a few days old.

The version you originally used (v0.3.0) does not yet support _Stand-In_ choice definitions.  Hence, the `?` identifier used in the latest demo project version was handled as a usual choice definition, resulting in the error messages.

Please always try to use a demo project version matching the KiVar release that you want to use it with.  Either from the corresponding [v0.3.0 Source Code Package](https://github.com/markh-de/KiVar/archive/refs/tags/v0.3.0.zip), or by browsing the [code of the appropriate v0.3.0 release tag](https://github.com/markh-de/KiVar/tree/v0.3.0/demo).

Thanks for explaining your observations.

markh-de, 2024-08-23:
Fixed on main branch.

Ref: 02cbfa6

closed 2024-08-23 by markh-de

Cocoader, 2024-08-23:
You are absolutely right - my bad. I did checkout **main** instead of the **v0.3.0** tag and used that example.

So if I want to use your latest (pre-release) version from **main** I can just generate it via ./[release.sh](https://github.com/markh-de/KiVar/blob/main/source/release.sh) _"insert latest version here"_ like you did with 
https://github.com/markh-de/KiVar/blob/a9a44e211d1a014fc2fc9780d3d1dca614311da1/source/kivar_backend.py#L23
> Well, I remembered wrong. The dialog already was modal, but I `Destroy()`ed the dialog from within an event handler. I've modified that part. Please install the following PCM Package using the [...]

until you finish your next milestone [v0.4.0](https://github.com/markh-de/KiVar/milestone/4) and release it in PCM?

markh-de, 2024-08-23:
Exactly.

I'm very short of spare time, but I hope to release v0.4.0 within the next days.  For the time being, you can use the `release.sh` script to build your own PCM packages.

Cocoader, 2024-08-23:
>I'm very short of spare time, but I hope to release v0.4.0 within the next days. For the time being, you can use the release.sh script to build your own PCM packages.

Please don't feel pressured - that wasn't my intention at all. I was more interested in your build pipeline - I don't know Python or how KiCad handles its plugins.

Your approach of collecting multiple issues / new features into a new milestone and create releases accordingly is the reasonable way to go. 

I just wanted to know if this the correct way to use your plugin in a sort of "nightly" manner without spamming the PCM with _maybe_ undocumented pre-releases like you described in #59.

Have a nice weekend and I'm more than happy with your fix in `0.3.9908` as is and knowing how to use your active development branch.

PS: I know that the Mac KiCad users are a very small subset in the great scheme of things hence I really appreciate your swift fix targeting those.