markh-de/KiVar #77: Ω not properly escaped somewhere [macOS]

Cocoader, 2024-09-16:
I'm not sure if the problem with the Ω is related to the bundled python version 3.9 of the macOS port of KiCad or inside the macOS port of KiCad itself. Can you reproduce this with python 3.9 in linux?

[image]

**Desktop (please complete the following information):**
- Version: 8.0.5, release build
- Platform: macOS Ventura Version 13.6.9 (Build 22G830), 64 bit, Little endian, wxMac
- KiVar 0.4.0

markh-de, 2024-09-16:
The encoding issue does appear when you directly load the unmodified demo project schematics?

And only in the custom fields ("Var"), not in the Value field?

Do I understand the issue correctly?

The original files are UTF-8 encoded.  How did you download the demo project?  Did you clone it with git, or did you use a release ZIP file?

As the rule fields are not touched by KiVar, I don't think that this is a KiVar issue, but more likely a Kicad/wxMac issue.

Can you please re-save the original demo schematics and PCB files (as you downloaded them) and attach them here, so that I can take a look at them and check if/how the encoding has changed.  Thanks!

Cocoader, 2024-09-16:
> The encoding issue does appear when you directly load the unmodified demo project schematics?

No, the schematic looks fine in the unmodified state - the issue appears when I update the Schematic from PCB. I just updated the PCB from Schematic once and added 2.5V(25) to VarID.Var of R34 because I think the pcb / schematic of your example 7 have been out of sync in your v0.4.0 release. Otherwise I get an error in KiVar. Can you reproduce this?

[image]

Everything looks fine in the KiVar window.

[image]

The issue appears when I want to Update Schematic from PCB.

[image]

> And only in the custom fields ("Var"), not in the Value field?

Correct.

Before:
[image]

After:

[image]

> The original files are UTF-8 encoded. How did you download the demo project? Did you clone it with git, or did you use a release ZIP file?

I did clone it and checked v0.4.0 out and tested it also with the archives .zip and .tar.gz of this release. The problem stays the same. 

> Can you please re-save the original demo schematics and PCB files (as you downloaded them) and attach them here, so that I can take a look at them and check if/how the encoding has changed. Thanks!

Sure this is https://github.com/markh-de/KiVar/archive/refs/tags/v0.4.0.zip without any changes and re-saved: [KiVar-0.4.0_re-saved.zip](https://github.com/user-attachments/files/17017436/KiVar-0.4.0_re-saved.zip). The schematic didn't change on save.

markh-de, 2024-09-16:
I believe this is a sync issue in KiCad.

Please try this project:
[issue77.zip](https://github.com/user-attachments/files/17019449/issue77.zip)

Please open the PCB and sync back the changes from the PCB to the schematic.  There are two fields that will be synced.  I noticed that the text seems to be interpreted as HTML.  Maybe this is part of the problem.

[image]

However, no wrong encoding appears in the schematic afterwards here on Ubuntu. 

If the encoding issue also shows up in this demo project, it is not a KiVar issue.  In this case, please consider filing a bug report in the KiCad issue tracker.

Thanks for pointing out the issues with the demo project.  I did a re-release that should fix the problems with the out-of-sync schematic and PCB.  Source code stays the same, but the tag moved by two commits, so that the correct demo project data is provided for v0.4.0.

Cocoader, 2024-09-16:
>I believe this is a sync issue in KiCad.

>Please try this project:
issue77.zip

>Please open the PCB and sync back the changes from the PCB to the schematic. There are two fields that will be synced. I noticed that the text seems to be interpreted as HTML. Maybe this is part of the problem.

>However, no wrong encoding appears in the schematic afterwards here on Ubuntu.

>If the encoding issue also shows up in this demo project, it is not a KiVar issue. In this case, please consider filing a bug report in the KiCad issue tracker.

I agree - this is a concise example which indeed _rules out KiVar_.

[image]

I will create a KiCad issue. Is it ok to reference this issue #77 and use your [issue77.zip](https://github.com/user-attachments/files/17019449/issue77.zip) as 'steps to reproduce'?

>Thanks for pointing out the issues with the demo project. I did a re-release that should fix the problems with the out-of-sync schematic and PCB. Source code stays the same, but the tag moved by two commits, so that the correct demo project data is provided for v0.4.0.

The re-release works as intended. I'm happy to help and wish you a nice talk at KiCon. I hope it will be recorded too.

Until this is fixed by KiCad I will stick with ASCII.

markh-de, 2024-09-20:
Sorry for letting you wait so long. Been busy with KiCon (yes, it was a great experience, and I believe they will provide videos).

> I will create a KiCad issue. Is it ok to reference this issue https://github.com/markh-de/KiVar/issues/77 and use your [issue77.zip](https://github.com/user-attachments/files/17019449/issue77.zip) as 'steps to reproduce'?

Yes, of course.  It might not be a very good reference, though ... but it should be good enough to illustrate the problem.

As we've now ruled out KiVar as the root cause, I'll close this ticket. Feel free to re-open it in case you disagree. :)

closed 2024-09-20 by markh-de

Cocoader, 2024-09-20:
>Sorry for letting you wait so long. Been busy with KiCon (yes, it was a great experience, and I believe they will provide videos).

Absolutely no problem - I figured you would be busy traveling / preparing. I'm curious about your talk.

>Yes, of course. It might not be a very good reference, though ... but it should be good enough to illustrate the problem.

Thank you. This should get the issue easily across without unnecessary complexity.

 >As we've now ruled out KiVar as the root cause, I'll close this ticket. Feel free to re-open it in case you disagree. :)

I agree.

markh-de, 2024-10-16:
The synchronization issue seems to be fixed in KiCad **8.0.6**: https://gitlab.com/kicad/code/kicad/-/issues/18406

The HTML interpretation persists, but this is a cosmetic issue only, not a functional one.

markh-de, 2024-10-16:
I just created an issue for the HTML topic: https://gitlab.com/kicad/code/kicad/-/issues/18925

Cocoader, 2024-10-16:
I did also created an issue https://gitlab.com/kicad/code/kicad/-/issues/18816 which was related to https://gitlab.com/kicad/code/kicad/-/issues/18406.

It now syncs correctly but fields which include non ASCII characters **aren't listed in the 'Changes To Be Applied' dialog** - at least in macOS. Maybe our issues are now somewhat related?

I do like your proposal https://gitlab.com/kicad/code/kicad/-/issues/18925 to list the reference designator in the schematic editor similar to the board editor.