markh-de/KiVar #79: Hide `Fab` layer for DNP components

Confucij, 2024-12-06:
When generating assembly drawings for different board variations, all DNP components stay in `Fab` layers, which can lead to confusion on the drawing. Is it possible to hide the `Fab` layer for DNP components?

markh-de, 2024-12-09:
Well ... it may be possible by moving all objects of DNP footprints from the _Fab_ layers over to some specific user layers (or back, that's why we cannot simply remove them), but tbh., I think your idea would make a good Feature Request for KiCad itself.

Because ideally, KiVar's task is only to switch attributes according to user-specified rules. Interpretation of the actual attribute states (in this case, hiding of DNP or Not-In-Pos components in plots) should be done by KiCad.

Also, as the SWIG interface is getting obsolete soon, I am tempted to avoid adding new features as I'm not sure about the new IPC API's capabilities and hence the level of compatibility we'll be able to maintain.

If you really need this feature quickly, you could implement it as a pre-processor filter script (PCB file in, PCB file out) that simply removes all DNP/not-in-pos components (or their _Fab_ layer objects, specifically) in a temporary PCB file that is used for the sole purpose of plotting the _Fab_ layers.

If you don't know how to do that, I can support you with that.

bshubenok-isr, 2024-12-16:
I got you. I will probably try to do this via PDF generation plugin

Confucij, 2024-12-16:
I got you. I will probably do this via PDF generation plugin.

PerennialNovice, 2025-05-30:
I just had the same idea, but with references on the silkscreen.
Especially when variants use different footprints on the same location.

Would that also be on the KiCAD side to handle visibility.

I'm not sure that it is wanted and expected behavior that the reference on silkscreen disappears for a DNP component?