markh-de/KiVar #81: Change entire component

Zuikkis, 2025-01-30:
**Is your feature request related to a problem? Please describe.**

Our company is transitioning from Altium to Kicad. We use our own component library, nothing from Kicad libraries. Our library has separate component for every different value, so for example RES1206_1K and RES1206_10K are different components, even though they share the same symbol and footprint.

We often need variations so different value component is used in place of other. This is not possible with KiVar because it only has access to parameter fields inside one component, you can't change entire component. Simply changing the Value field is not enough as our manufacturing works entirely based on component names, RES1206_1K with Value=10K will be mounted as 1K.

**Describe the solution you'd like**
Change component. The KiVar properties would need to be copied over to the new component, I guess? Other fields could be taken from library.

**Describe alternatives you've considered**
I could use separate "Mounted name" -field that originally is copied from component name, but then KiVar could change that field. When processing files to our manufacturing, use that field instead of component name. But this can be a source of confusion if component "RES1206_1K" has "Mounted name" of "RES1206_10K". However this would work with current KiVar.

Another possibility is to add two components in the schematic and place them over each other in the PCB. Then just populate only one of them. But this makes the schematic look messy and causes DRC errors in the PCB.

**Additional context**

Zuikkis, 2025-01-30:
After inspecting this further, it seems we perhaps need to store our component name in "Value" field anyway. KiCad BOM and Pick&Place generators aren't very customizable and there seems to be no way to get component name included in the listings, but Value is there..

So perhaps I'll stick with my "alternate method" described above.

I'll close this. :)

closed 2025-01-30 by Zuikkis

markh-de, 2025-01-30:
This is a known issue, also addressed during my KiCon talk: https://youtu.be/SpXH380KWUA?t=2665

Modifying the Database Library symbol reference is not possible with the pcbnew SWIG interface that KiVar is (still) using.  When the new IPC API will be available for eeschema (the schematic editor), this might become possible and may be implemented in KiVar.  The database reference (unique part identifier) could then be modified by KiVar, just as any other field.  It may be required to manually update all Database Library symbols afterwards, so that their field values are updated accordingly - or maybe this can also be done through the API ... we'll see.

markh-de, 2025-01-30:
If you cannot output the component ID to your BoM, can you maybe modify your component library to contain that ID in a dedicated field that is transferred to your KiCad symbol instances?

For example, if you use a REST API server (instead of a database), you could simply mirror that field to the corresponding REST response, have it stored in a symbol field and get it exported to your BoM from there.

Also, consider creating a feature request for exporting component IDs in the KiCad issue tracker.