markh-de/KiVar #83: Support for "Sub" Varients

bluespider42, 2025-02-06:
Is it possible to define choices that apply to variants that match a pattern, eg. start with the same letters (or by some other means)?

eg. I have 4 variants (but really 2 main variants, each with 2 sub-variants) so a component might have `Var: Variant A_1(10k) A_2(12k) B_1(20k) B_2(22k)` but for most components the only difference is between 'A_x' and 'B_x'. 

So ideally I'd like to be able to define them something like `Var: Variant A_*(1uF) B_*(1nF)`. 
This also means that if I ever add an option for 'A_3' I only have to make the change in the places that really matter, and not change every component. Currently they would look like `Var: Variant A_1(1uF) A_2(1uF) A_3(1uF)  B_1(1nF) B_2(1nF)`. 

Up to now I have only been dealing with the case where I have 'A_1, A_2, and B' So I can use the default choice for the A_x, but this makes it less explicit about the intention and doesn't allow for any 'B' sub-variants.

This could even be expanded to fit with the current '*' and '?' pattern `Var: Variant A_?(10k) B_?(100k) A_5(13k) B_3(130k)`.

markh-de, 2025-02-06:
This feature is currently not planned.  I also thought about introducing placeholder names, but have decided not to implement it for the following reason:

One of the concepts of KiVar is that Choice definitions (in component fields) shall be independent from each other.  Using placeholders would create dependencies to other definitions that mention the (full) Choice name that the placeholder string is referring to.  The idea is that you have any number of components that declare or refer to any number of _definitive_ Choice names, without depending on each other.

Default Choices (`*`) and Stand-In Choices (`?`) are a bit special.  They only match for "unmentioned" ("else") cases, not for a _specific_ set of matching Choice names.

But yes, you could argue that Default and Stand-In Choices would only need to be extended by a matching algorithm (incl. error checking).  And I'm open to implementation suggestions (PRs).  But currently I simply don't have the spare time to do it myself.  Sorry.

For your example `Var: Variant A_1(1uF) A_2(1uF) A_3(1uF)  B_1(1nF) B_2(1nF)` I would recommend using a combination of Stand-Ins and Choice name grouping, e.g. `Var: Variant ?(1uF) B_1,B_2(1nF)`. I think this is not too bad to use, and you keep your rules independent ... Well, you'd need to decide whether and where to use the "`?`" Stand-In Choice if you really want to be independent from other Choice name declarations ... The idea is that if you remove a key component that held the declaration(s) of one or more _definitive_ (no placeholder contained) Choice names, then all your other components still have valid rule definitions.

bluespider42, 2025-02-07:
Hi, thanks for the response - that's understandable. I hadn't actually realised the `B_1,B_2(1nF)` type notation which actually helps a lot with keeping things readable and maintainable.
If I added a `B_3` variant I could just do a text search and replace for example.

markh-de, 2025-02-09:
If you know that a Choice definition shall be the default case for any Choices added later, you can also add the `?` (Stand-In) Choice, e.g. `B_1,B_2,?(1nF)`, to that definition.