markh-de/KiVar #84: Control KiVar from another action plugin

mattes34, 2025-02-27:
First of all, thank you for KiVar!
I want to create a small action plugin to export gerber/bom.. , but how can I handle it with KiVar ?
Is there a way now to control KiVar from another action plugin ? ( maybe by dirty copy of Kivar_backend.py ?
Using the kivar-cli is not possible due to the opened board or ?

markh-de, 2025-02-28:
Although KiVar is not designed to be integrated in other plugins and does not have a public API, I think that copying the backend module to your plugin code directory (preferably without changes, so that updating it stays simple) and calling its functions from your own plugin would work.

For example code that shows you how to use the module in an Action Plugin, I'd recommend you have a look at the plugin code. At least for the init stuff, such as creating the internal dicts from a board that you got from pcbnew's GetBoard() (not from a file, as you correctly pointed out) then passing those dicts around.

For the actual variation setting stuff (assigning Choices to Aspects, I guess that's what you want to do), I think the CLI code is a cleaner reference.

I just tried setting up a jobset (new KiCad 9 feature) that calls KiVar (CLI) before it exports the data. Unfortunately, the project's schematic and PCB are opened in the editors just before the job (i.e. KiVar) is run and any exported data is then taken from these open instances. So the changes that KiVar does in the PCB file are not reflected in the export. The jobset feature feels more like a "remote control" that commands the open editors. Should KiVar ever move to the new IPC API, then it should be able to perform the variant switching directly in the open editors just before the data gets exported. I'm not sure if this sounds like a clean design, but it should be possible.

Personally, I prefer exporting data (after automatically panelizing the design) using KiKit from a shell script. This approach is very flexible and allows for switching variants just before the export. We'll see how this will change when the SWIG interface is removed.

markh-de, 2025-02-28:
As modifying the board from within a jobset would also enable other nice use-cases (e.g. automatically generating board renders for different variants, even from within one and the same jobset), I think being able to reload the board from file (disk) might actually be a useful feature.

I just created a _New Feature_ issue [\#20162](https://gitlab.com/kicad/code/kicad/-/issues/20162) for KiCad.

Let's see what the devs think about it.

mattes34, 2025-05-08:
Hi, thanks for your reply.
I was trying to find a simple solution to build the output files in a fixed format and a simple why to publish the script internally to our developer on Windows and no server side scripting.
I am now getting it via the CLI. But it also feels creepy due to the windows path of the kivar cli.

The jobset sounds a nice why, but yes, you write it.
Hopefully kicad will get a native variant support.

markh-de, 2025-05-08:
> But it also feels creepy due to the windows path of the kivar cli.

Maybe just add something like `C:\Users\foobar\Documents\KiCad\9.0\3rdparty\Python311\Scripts\kivar.exe` to your `%PATH%` variable? I'm not very familiar with Windows, but I think this would be quite clean solution, wouldn't it?