markh-de/KiVar #85: Single variant support for combining sets of aspect choice assignments

markh-de, 2025-04-25:
This will allow KiVar to be used like classic variant management tools.

There will be a single variant that can bind multiple aspects (free aspects still allowed).

Variant management (binding of aspects, assigning choices for each variant) will be done in an external CSV file to allow seperation of concerns (developer vs. manager view).

bluespider42, 2025-04-30:
Might be worth considering a yaml file in the kibot style to hopefully ease future integration rather than a csv file with a new way of specifying kicad variants.

markh-de, 2025-04-30:
I'll need to look into the file format you mentioned. However, as we have the Aspect (scope) layer, I doubt that this would be compatible. In KiVar, variants only specify the Choices required for the set of (Bound, i.e. variant-attached) Aspects.

If it's possible to support it in a compatible way, we should consider adding it later as another alternative file format (there may well be several).
I'm planning to release version 0.5.0 within the next few days.

markh-de, 2025-04-30:
Just to explain how variants are defined in KiVar:

 * https://github.com/markh-de/KiVar/blob/single-variant-support/README.md#mixed-approach
 * https://github.com/markh-de/KiVar/blob/single-variant-support/README.md#top-level-variants

_(Note: links only valid until that dev branch is merged)_

bluespider42, 2025-04-30:
Ok - I hadn't realised it would be that imminent!
I am currently using Kivar as it's the best fit for my projects. The main draw back for me is automatically generating all the outputs.  What is needed is variant support as suggested in this issue, the ability to back annotate the schematic from the cli (I am aware there is an open kicad issue for this), and for kibot to understand these variants to sensibly generate outputs with appropriate file names etc.

This is just a suggestion - but could be a way of including the relevant info into kibot config - the only change from the current kibot format is the aspects key.  Currently this would result in an error `Unknown variant type: `kivar` from kibot, but could be something to have in mind for the future.

```yaml
variants:
  - name: 'AssemblyA'
    comment: 'Description of A'
    type: kivar
    file_id: _A
    variant: assembly_a
    aspects:
      - aspect1: 1a
      - aspect2: 2a

  - name: 'AssemblyB'
    comment: 'Description of B'
    type: kivar
    file_id: _B
    variant: assembly_b
    aspects:
      - aspect1: 1b
      - aspect2: 2b
```

markh-de, 2025-05-03:
Relates #82

closed 2025-05-03 by markh-de