markh-de/KiVar #88: SyntaxError: f-string: expecting '}' on Kivar-cli 0.5.0

mattes34, 2025-05-08:
**Describe the bug**
I get an syntax error with the kivar-cli

**To Reproduce**
Steps to reproduce the behavior:
1. KiCad 9.0.1
2. Kivar 0.5.0
3. Kivar 0.5.0 cli via pip
3. Kicad command promd
  kivar -h
`
kivar_cli.py", line 335
    Msg().color('pass').text('Check passed.').reset().text(f'  Matching {'variant and ' if check_variant_match else ''}choices found for complete set of {len(sel)} aspect(s).').flush()
                                                                          ^^^^^^^
SyntaxError: f-string: expecting '}'
`

**Desktop (please complete the following information):**
 - OS: Windows 11

** fix ** 
Using " instead of ' in the string
`
f"  Matching {'variant and ' if check_variant_match else ''}choices found for complete set of {len(sel)} aspect(s)."
`

markh-de, 2025-05-08:
Thanks for reporting this.

Very interesting! I just learned that Python 3.12 changed the way f-strings are interpreted. Reading my code again, I really wouldn't have expected it to work in the first place. However, it did (I'm sure I tested it) ... but only on my Linux system, where I have distribution-provided Python 3.12.3 instead of KiCad-provided Python 3.11.5 (as on your Windows system).

I'll fix that asap.

closed 2025-05-08 by markh-de

markh-de, 2025-05-08:
Just for reference, you can try this on Python **< 3.12** vs. **>= 3.12**:

```
print(f'hello {'foo' if True else 'bar'} world')
```

markh-de, 2025-05-08:
It's released and available on PyPI:

https://pypi.org/project/kivar/

Please `--upgrade` and test. Thank you!

mattes34, 2025-05-08:
<3.12 not working
>= 3.12 working (PEP 701 – Syntactic formalization of f-strings)

last kivar pip installation is working
Thanks!

commit 254840c3 (2025-05-08): Fix nested quoting in f-strings (for Python < 3.12)

Fixes #88