markh-de/KiVar #92: Is there a way to have different variant choices in different instances of the same sub-sheet in a schematic heirarchy?

colbyboles, 2025-05-30:
I have a design that sometimes has two CAN controllers. I have a single sheet (CAN.kicad_sch) that I use twice within my parent schematic sheet to create the two CAN controllers. In some cases I only want to build a board with one controller populated. 

If I create variants and make choices using the Var fields of the components within CAN.kicad_sch, they take affect, but unfortunately both instances behave exactly the same regardless of which one you edit. If I DNP one CAN controller design, the other one becomes DNP as well.

It seems like what is needed is a way to DNP an entire sheet within a parent design, but I can't see how Kivar could do that since it seems to work in the component field level and there isn't something equivalent for sheets.

Is there perhaps some workaround or trick to doing this? I'd really prefer to keep using sheet heirarchies like this to maximise design re-use instead of flattening everything out, which I know could allow me to use Kivar as-is.

markh-de, 2025-05-30:
Please see [my reply](https://github.com/markh-de/KiVar/issues/86#issuecomment-2908106238) in issue #86.

This is an issue in KiCad which cannot be resolved by a third-party plugin.

The way KiCad synchronizes the PCB back to the schematic has potential for improvement regarding variables.

Please consider creating an issue for KiCad.

colbyboles, 2025-05-31:
Thanks for the reply - I will "flatten" these two CAN controllers into a single subsheet for now and create a KiCad issue, or find an existing one to add on to and hope for the future.

closed 2025-05-31 by colbyboles

markh-de, 2025-05-31:
Feel free to post a link to that KiCad issue here, as I may have an idea how the problem with baked values could be solved or at least worked around.